freakified/TimeStylePebble #10: Disable weather update during sleeping time?

jemyzhang, 2015-07-23:
From code:
```
  // every 30 minutes, request new weather data
  if(!Settings_disableWeather) {
    if(tick_time->tm_min % 30 == 0) {
      messaging_requestNewWeatherData();
    }
  }

```
Is that a good idea to mute the weather update during sleeping time, for example, 00:00 - 5:00, to reduce the battery consumption? 
Or more complex, add some option on phone settings page.

tilden, 2015-07-23:
It's possible, but personally I haven't observed any major battery drain problems caused by the weather updates at night.

jemyzhang, 2015-07-24:
Hi, I am not means the battery drain, just a so-called optimization ^^

notlaforge, 2016-02-18:
I think there's an API call now that tells you if the person is sleeping or not.

tilden, 2016-02-20:
Yes, it's how it currently determines whether or not to show the sleeping icon. Honestly, I'm not sure if this optimization is worthwhile, so I'm going to close the issue for now unless someone has strong objections.

closed 2016-02-20 by tilden