freakified/TimeStylePebble #11: Show battery percent when charging

jmarhold, 2015-07-23:
The current value of battery when charging is weird. E. g. watchface shows 90%, then i turn on charging and watchface shows 70%.Maybe add condition to  line 145 of main.c : if(Settings_showBatteryPct && !chargeState.is_charging) {}

tilden, 2015-07-24:
I'm not sure what you're saying. That the percentage should always show when charging, regardless of the setting?

jmarhold, 2015-07-24:
I am just saying that percentage has different value before and while charging. See photos - 90% vs 60%.
[image]
[image]

Its little bit confusing,so i suppose to not show percentage while charging (pebble in settings also doesn't show percentage, only charging icon)

tilden, 2015-07-24:
Oh, thanks for the photos, now I get it. I just verified it on my watch as well -- the Pebble Time really does show strange numbers when the watch is charging. However, I also tested on an original Pebble, and the charge numbers make sense.

Because of that, it looks like this is really a bug in the Pebble Time's firmware. I [posted on the developer forum](http://forums.getpebble.com/discussion/27918/battery-charge-percent-returning-crazy-numbers-when-charging-but-only-on-the-pebble-time) -- maybe someone from Pebble will see it and comment.

tilden, 2015-07-24:
Issue fixed in v2.3

closed 2015-07-24 by tilden