freakified/TimeStylePebble #15: Hide the weather icon if it gets too out of date

tilden, 2015-08-19:
If the watchface hasn't communicated with the phone in a long time, it will continue to display the last weather information that it received, whatever it was.

A better approach might be to hide the weather icon after a its age exceeds a certain amount (a day? 12 hours?)

jgallen23, 2015-11-16:
I like this idea. Sometimes I don't feel confident in the weather getting updated, so I leave the watchface and come back. It would be nice if grayed out or disappeared if it didn't update in the last 1 or 2 hours.

superm1, 2016-04-06:
Another way that would be useful would be showing in really small text underneath the weather what time it was last updated.  not sure if there's enough room for this though

tilden, 2016-04-06:
Definitely not enough room, I'd say.