freakified/TimeStylePebble #19: Week number option in sidebar

tilden, 2015-08-19:
If issue #16 is completed, there would be new possibilities for what could be displayed in the sidebar.

I've had a request for a current week number to be one such option.

wicol, 2015-09-18:
I would love to see this. I keep switching between this and "TIMEWATCH" which has a nice icon for precisely this. Check it out for inspiration :)

tilden, 2015-09-18:
It's actually coming in the next version! :sunglasses: 

Question for you: in your country, how do you expect to see week numbers shown? Just a number? The local word for "week" and the number? Something else?

wicol, 2015-09-18:
Awesome! Personally I dont mind english names of days/weeks/months as long as I can get something like ISO 8601. But W38 is usually written as V38 in swedish since Week = Vecka. Looking forward to next release!

tilden, 2015-11-19:
Completed in fb21b3d4e97f0fd3eb2effa808ef1dc03be96ec1

closed 2015-11-19 by tilden