freakified/TimeStylePebble #21: Persistent data may randomly reset on iOS devices

tilden, 2015-08-23:
I have reports of the persistent settings resetting randomly on Pebble Time watches paired with iOS devices, however I have been unable to duplicate the issue.

sappling, 2015-08-30:
Not sure if this is the same issue.  I'm using iOS, but my settings never seem to save. If I change the theme to "Pure Dark" and hit "Send to watch", the changes apply, but if I use the watch buttons to change to another watchface and then switch back, the settings are all back to default.  This happens every time.  I'm using a classic pebble (not pebble time).

tilden, 2015-08-30:
Interesting, you're the first person who has reported this on the OG Pebble, but it definitely sounds like it could be related. Do you have access to the SDK?

sappling, 2015-08-30:
I had the SDK installed about 9 months ago, but it's on an old version and not on my primary computer.  I can probably get it back, but not easily.  Is there a way I can help you debug without this?

tilden, 2015-09-13:
After a lot of research, I came up with a guess at the cause of this bug. I fixed what I believe was causing it, so I'm going to close this issue for now in hopes that it is resolved.

closed 2015-09-13 by tilden