freakified/TimeStylePebble #24: Enter animation for sidebar

TomerRosenfeld, 2015-08-29:
It would be really neat if there could be an entering animation for the sidebar like this one:
[image]

tilden, 2015-08-30:
How would you envision the animations going?

The animation on the Timeline itself are always traveling up or down, based on the direction in the timeline that you're going. For the watchface, you're in the present, so we can't really say up or down. What do you think?

TomerRosenfeld, 2015-08-30:
I meant for only entering animation, so when the watchface is loaded the taskbar will make an animation from bottom to top kind of like the weather icon in the gif

tilden, 2016-02-16:
Especially now that Pebble has change the watchface entry animation so that it just slides in, I don't think this is needed any longer

closed 2016-02-16 by tilden