freakified/TimeStylePebble #35: Time widget to the sidebar

derrij, 2015-12-10:
Please add Time widget  to the sidebar. 
It adds ability to read time under the cuff.
Thank you.
[image]

Jajunk, 2016-01-30:
I was able to add this. I did run into compile errors right off the bat with the master branch though. Maybe the owner can comment on this error message: 

../src/clock_digit.c:131:22: error: invalid operands to binary == (have 'GColor' and 'GColor8')

There are a few of these existing from a fresh pull of master.

I am able to build fine if I comment those lines out (they all involve checks for APLITE watches).
Also, I have no idea how to modify the settings page to add the widget as an option to the Complications section, so I hard coded it to test.

[image]

tilden, 2016-02-01:
Awesome! Yeah, the Aplite related errors are caused by the changes for black and white watches in SDK 3.8. I've already fixed them in the other branch, but I need to finish adding grey support before I merge it to master and release.

I'm hoping to get that version pushed by this week, then I can work on merging your code. Go ahead and make a pull request!

tilden, 2016-02-09:
This has been merged and added to the 4.8 release. Thanks for the contribution!

closed 2016-02-09 by tilden