freakified/TimeStylePebble #37: Phone battery indicator

lordyama, 2016-01-18:
This is my favourite watchface! It works very well with the Pebble Time UI.

The one thing missing for me is a well to tell the battery status of the connected smartphone (Android). This will save me from having to periodically check my phone (which itself drains the battery). The watch itself has great battery life, so I don't need to know how much power is in it.

I understand that the Pebble app doesn't report phone battery status, so this will require a companion app. Some apps already do this (e.g. [Glance] (http://www.finebyte.co.uk/) and [Canvas] (https://play.google.com/store/apps/details?id=com.pennas.pebblecanvas)). I used to use Glance, but I'd prefer a watchface so that I can make full use of the timeline interface.

tilden, 2016-02-12:
This can't be done without a companion app, as you say. I personally consider this out of scope for the project, since it would add an enormous amount of complexity for a single data point. That said, though, if someone else contributed such a feature I would consider integrating it.

tilden, 2016-03-10:
It turns out this *may* be doable using the technique from here https://github.com/aHcVolle/TrekVolle/blob/master/src/js/battery.js#L87 More analysis is required to determine which phones/OSs this works on.

OPerepadia, 2026-11-06:
I'd like to try implementing this. It seems there is a new [plugin API](https://developer.repebble.com/guides/plugins/preview/) that has a built-in `phone/phone_state` source, so no companion app is needed. This should work on both Android and iOS.

tilden, 2026-11-10:
I'm still thinking about how I want to approach plugins in general, since there's a lot more than just phone state. 

I was currently thinking:
- I don't want to add any plugin-specific features until plugins are rolled out for 100% of users
- I'd prefer doing an all-in-one "plugins are now supported" release than just each plugin feature piecemeal

So that's currently my plan!