freakified/TimeStylePebble #38: Forecast.io API support

bbs4us, 2016-01-20:
It will be great if someone can add support for [The Dark Sky Forecast API](https://developer.forecast.io/docs/v2). API calls are limited but we can use our own API keys.

tilden, 2016-01-20:
If someone is interested in adding this I would definitely accept the pull request.

bbs4us, 2016-01-21:
I'll have a try (:

Technowix, 2016-02-12:
I want to try too :)

bumper314, 2016-05-21:
Just wanted to see if anyone has made progress on this before I spec out this job for a contractor.

tilden, 2016-05-22:
I've made some progress on my local branch, I'm hoping it have it released this week.

tilden, 2016-05-29:
Weather underground support was added in version 6.0, which I believe should cover underlying issue (that OpenWeatherMap data is terrible.) If there is significant enough demand, this issue could be reopened.

closed 2016-05-29 by tilden