freakified/TimeStylePebble #46: Light weight LECO option

tilden, 2016-02-16:
Now that there are bold/light options for the default font (Avenir), LECO could conceivably have the same options

foscoj, 2016-11-27:
This issue is exactly the reason why I was browsing the repo over here, trying to understand the concept of pebble watchfaces and especially yours, which has been my default from day one with my pebble2.
Unfortunately without the weatherCommon (weather) and an example structure of the secrets file, the project won't compile in cloudpebble (being abroad, no own machine to tinker with the sdk).

Judging from the "extra" portion of code for the LECO setting, the LECO "only minutes bold" Setting will need a bit debugging, would love to help you out there.

Could you help me out with a "dummy" file for weather icons and secrets, so I'm able to debug the LECO "only minutes bold" setting?

foscoj, 2016-11-27:
OK, tried to play around with this task and got stuck on the font creation for the fctx lib, missing the LECO svg :-( 

How did you create the other ones? Am I missing a download somewhere on pebbles page?

tilden, 2016-11-28:
It previously used images to display the numbers, but I needed to switch to FCTX in order to add Quick View support, which required the ability to scale the time up and down.

In order to create fonts, you need to use [pebble-fctx-compiler](https://github.com/jrmobley/pebble-fctx-compiler) to convert the SVG fonts into FCTX's "ffont" format. You'll probably also need to edit the SVG fonts themselves to make the digits all the same width (large-digit faces tend to look weird with varying character widths). In the case of non-thin LECO, I needed to keep recompiling the font and tweaking the metrics until it looked okay.

foscoj, 2016-11-28:
Where do I Get the thon leco svg?

tilden, 2016-11-28:
I got it from MyFonts, but you're right that the lightweight version isn't on there.

tilden, 2026-11-03:
Added in 8.0.0!

closed 2026-11-03 by tilden