freakified/TimeStylePebble #50: Sidebar widget: Day Number

tilden, 2016-02-17:
In some industries, people rely on the day number (1-366). This can be obtained via a standard call to strftime, so this is a prime target for a new sidebar option.

The most difficult part will be adding the word for "day" in 29 languages.

jmarhold, 2016-02-27:
Yes, it is true, the most difficult part is adding the word for "day". ;-) As not a native speaker of most of languages i really do not know how to abbreviate for example italian "giorno" or greek version of this word, etc. I think that it would be nice to have for each language some guy, say guarantee that will be responsible for some language. What is your opinion?

bumper314, 2016-05-31:
The BEAT internet time uses an @ symbol.  What if we just use a # symbol for Day number?

tilden, 2016-05-31:
It's definitely possible, but it wouldn't match how it currently display week number. Then again, day number is a niche feature so perhaps it wouldn't matter too much.

Another key obstacle: any new widgets going forward will not work on classic Pebbles, because I'm using nearly all the available RAM.