freakified/TimeStylePebble #55: Support alternate time zones with 0:30 offset (India, Venezuela, etc…)

mwarrenus, 2016-02-19:
[1.5 billion people](http://www.wolframalpha.com/input/?i=population+of+india+%2B+population+of+venezuela+%2B+population+of+myanmar+%2B+population+of+iran+%2B+population+of+afghanistan) live in areas with [time zones offset by 30 minutes](http://www.timeanddate.com/time/map/). Please allow the alternate time zone complication to support half hour offsets.  Thanks!

tilden, 2016-02-23:
This is doable, I think. The hardest part will be figuring out how to fit the whole time onto the sidebar.

In addition, now that we've moved to OS 3.x on all Pebbles, we can use `gmtime` instead of just local offsets. This would allow users to just pick a timezone instead of local offsets like we do now. (Disadvantage: we would need to reset everyone's time zone setting, so hopefully we wouldn't upset too many people)

mwarrenus, 2016-06-24:
I would be OK with the reset.

tilden, 2016-06-24:
Don't worry I'll get to this feature eventually :smiley:

tilden, 2026-11-03:
Released in 8.0.0 -- it now uses the phone's timezone database directly, which should solve all of the timezone issues!

closed 2026-11-03 by tilden