freakified/TimeStylePebble #65: Hide the disconnected device icon

marcocipriani, 2016-03-17:
My favorite configuration of the TimeStyle watchface is the one with the battery percentage, the calendar and seconds on the sidebar (in this order, from top to bottom).
Sometimes when the phone is connected to my Pebble (also to the LE mode), the watchface doesn't recognize it and on the screen there's **the icon of the disconnected device instead of the calendar**.
Especially when my phone is out of battery or I want to use my watch stand alone, I'd like to quickly remember myself what day is.
I know it's a smartwatch, so it's meant to be connected with my phone, but that icon is not always useful.
Is there a way to hide it? If not, I hope it could be a feature of the next version :)

plarus, 2016-12-03:
I have done this correction.
I is not a big change. But if you (@freakified) are interested: #97 #98

tilden, 2016-12-05:
Thanks for the update! Should be in the latest release

closed 2016-12-05 by tilden