freakified/TimeStylePebble #70: Add sleep mood / power saving mode

livekn, 2016-04-14:
It can be automatic activated by time. When it's enabled, no hourly vibrate (even if it was turned on), no weather update check, no seconds, ignore bluetooth state.

Coollision, 2016-06-11:
I'd love to see this function happen as wel! This  way i don't hav turn of hourly vibrations of at night because somtimes they wake me up.....

tilden, 2016-06-11:
Yes, there have been several requests for some kind of quiet time. I'm hoping that Pebble releases a Quiet Time API or just disables vibrations for 3rd party apps during quiet time. Their current implementation (notifications only) doesn't make sense, IMO.

mvoropaiev, 2016-10-12:
Would like to see such option as well, there are some other watchfaces allowing to set the start and end times for quiet time, but I really like this one :smile: So with this approach you at least can set it to match scheduled quiet time on pebble, which is kinda okay for night times.

tilden, 2016-11-25:
Pebble has finally added a quiet time API, and as of version 6.11 the watchface respect those settings!

closed 2016-11-25 by tilden

mvoropaiev, 2016-11-25:
@tilden awesome news, thanks!