freakified/TimeStylePebble #71: Forecast showing same temperature for min and max

ModusVivendi, 2016-05-11:
The forecast complication has been showing the same minimum and max temperature for quite a while now.  For clarification, its not that the forecast temperatures doesnt update, but that everytime it updates the same two numbers are shown from the minimum and maximum.

tilden, 2016-05-12:
Yeah, this is a definite issue. The problem is, OpenWeatherMap doesn't actually offer daily forecasts for free, so I'm actually just pulling the next 24 hours of hourly forecasts and using the min/max temps. It's possible that if I use the next 48 hours this issue will happen less.

ModusVivendi, 2016-05-13:
The two numbers have been showing the same temperature for at least the past month or more.  I have never seen it show different numbers in the min and max during this time.  If it was correctly pulling hourly temps for the next 24 hours, I would expect the two numbers to be different at least some of the time, but its never different.

tilden, 2016-05-29:
Fixed in version 6.0

closed 2016-05-29 by tilden