freakified/TimeStylePebble #74: Add combined date/week number

machineboycom, 2016-06-02:
It would be nice to have a possibility to add the current date (maybe without month or day?) to the week number complication.

Mockup: http://imgur.com/lueWhad

[image]

tilden, 2016-06-10:
It's interesting, but kind of a niche case. A more general approach would be to split out components of the date element into separate parts.

In addition, I suspect that "W23" would have trouble fitting if displayed using the canonical Pebble fonts.

dagbdagb, 2016-08-03:
Side note:
Also note that one should not trust week numbers  across geographical locations. 
See https://www.msoutlook.info/question/week-numbers-calendar-wrong.

tilden, 2016-08-03:
Closing for now; I believe a more general approach to widget compaction would be preferable to just implementing special combined widgets.

closed 2016-08-03 by tilden