freakified/TimeStylePebble #77: Weather data not updated (regular) using weather underground

MariusRumpf, 2016-06-15:
I have set Location to Automatic (GPS), Units to Celsius, Data Source to Weather Underground and entered my API key. Looking at the api analytics there where very few calls to the api. Maybe only those when submitting the watchface settings. When setting the Location to Manual (Hamburg, Germany) I have no updates at all, not even when submitting the settings to the clock.

Here are the api usage metrics:
[image]

Might be related to #76.

tilden, 2016-06-15:
Strange indeed, I'll have to look into this. Are you using "forecast", "current conditions", or both?

MariusRumpf, 2016-06-15:
Weather: Current only, selected in top Complications.

tilden, 2016-06-15:
One more question: is the exact location string "Hamburg, Germany"?

MariusRumpf, 2016-06-15:
No sorry, it is `Hamburg, Deutschland` as selected from the Google Autocomplete Suggestion.

MariusRumpf, 2016-06-15:
If you have an idea how I might help you with debugging feel free to tell me.

stevier5000, 2016-06-17:
Having a similar (same?) issue. I just signed up for a weatherunderground API key yesterday, and only saw 2 API calls in the last 12 hours. For instance, right now its 8:30am local time, and my current weather still shows a "moon" icon, and "..." for the forecast. Location string is "Huntington Beach, CA, USA" for me. I have current weather on the top, forecast in the middle.

MariusRumpf, 2016-06-17:
Debugged it just now and received the following errors
```
[19:46:08] javascript> TimeStyle:652 Get weather function called! DisableWeather is 'no'
[19:46:08] javascript> TimeStyle:1038 http://api.wunderground.com/api/API_KEY/forecast/q/Hamburg%2C%20Deutschland.json
[19:46:08] javascript> TimeStyle:1007 Uncaught TypeError: Cannot read property 'temp_c' of undefined
[19:46:08] javascript> TimeStyle:1045 Uncaught TypeError: Cannot read property 'simpleforecast' of undefined
```

stevier5000, 2016-06-17:
Looks like maybe the autocomplete locations are leading to the wrong query. Based on Marius' comment above, its translating the location into a URI. Presumably, mine would have been:
http://api.wunderground.com/api/APIKEY/forecast/q/Huntington%20Beach%2C%20CA%2C%20USA.json
which doesn't work - I get json back for every "Huntington" across the US. This is the proper url:
http://api.wunderground.com/api/APIKEY/conditions/q/CA/Huntington_Beach.json

I tried entering "CA/Huntington_Beach" in the watchface settings, but doesn't appear to be working. Not sure how to debug and dig in further.

tilden, 2016-06-18:
Thanks for looking at the logs! Yes, it appears that the issue is if Weather Underground is having trouble disambiguating certain locations. I'll have to see if I can find a way to make it more decisive.

MariusRumpf, 2016-06-18:
As a quick fix I put my geo location in and did not choose a location from the auto suggestion.
A good fix might be to save the geo location you get from the google suggestion selection and send that to the api. Does this work with open weather?

tilden, 2016-06-20:
I just pushed out a release with a fix. The approach I used ended up being essentially what you proposed; saving the geo location from Google. It's also able to fall back to the original method so that people with currently working weather won't suddenly find it broken, and users in China should still be able to input location strings.

closed 2016-06-20 by tilden

MariusRumpf, 2016-06-20:
Thanks a lot for the bugfix and the great watchface!