freakified/TimeStylePebble #78: Geolocation weather doesn't work

chibigoku04, 2016-06-20:
**Watch**: Pebble Time Steel
**Phone**: Nexus 6P (64GB, Android N NPD56N Dev Preview 4)
**Watch Face Version**: 6.3 (updated June 19th)

When Weather location is set to Automatic (GPS) the watch face doesn't update the current weather. When Weather location is set to Manual, the watch face does update. I have tried setting it back to Automatic after the initial retrieval and the watch face uses the "cached" Manual location values.

livekn, 2016-06-21:
I also have this problem. The error from wunderground is "querynotfound", location is /api/(my_key)/conditions/q/undefined,undefined.json

tilden, 2016-06-21:
Hmm, it would appears that the last version may have broken something with weather when automatic location is used. I'll have to look into it and see what I can find.

tilden, 2016-06-22:
Allright, I just pushed out a new version that should fix this issue. The problem was that the logic I added for the stored coordinates was running even when you _didn't_ have stored coordinates.

closed 2016-06-22 by tilden