freakified/TimeStylePebble #79: Celsius setting doesn't take effect in "Weather: Forecast"

QuanticMonkey, 2016-06-21:
**Watch**: Pebble Time Steel - FW 3.13 (latest)
**Phone**: iPhone 5S 64GB - iOS 9.3.2 (latest)
**TimeStyle**: 6.3 (latest)
**API**: OpenWeather and WeatherUnderground (same issue with both providers)

The "Weather: Forecast" field still displays temperatures in Fahrenheit although the "Units" option is correctly set to Celsius in the app (and has been successfully transmitted to the watch).

Although I only report it now, this bug is present for at least 3 months.

PS: thanks for this great app!

Cheers!

tilden, 2016-06-22:
Not sure what you mean; it's working perfectly on my end, and I haven't received any other reports of this problem (I have enough non-US users that if Celsius mode didn't work I'd be getting a steady stream of complaints!)

I just tested it, and there were no issues; unlike some faces, conversion occurs on the watch itself, so no refresh is required in order to do unit changes.

Default setting (fahrenheit)
[image]

Switched to celsius:
[image]

tilden, 2016-08-03:
Closing for now due to lack of confirmation.

closed 2016-08-03 by tilden