freakified/TimeStylePebble #83: Add Outlook week numbers

Muffo, 2016-08-12:
This is a great watch-face!

I really like the idea to add the week number because in my office we use it a lot to mark deadlines instead of the day and month.

One thing that I noticed is that the watch-face displays a week number that is different from Outlook (which we use in the company).
According to the watch the current week is 32, but Outlook says that it's 33.

[image]

I was looking at the code and I have seen that the week number is calculated using the function `strftime`: http://man7.org/linux/man-pages/man3/strftime.3.html

```
       %U     The week number of the current year as a decimal number, range
              00 to 53, starting with the first Sunday as the first day of
              week 01.  See also %V and %W.  (Calculated from tm_yday and
              tm_wday.)

       %V     The ISO 8601 week number (see NOTES) of the current year as a
              decimal number, range 01 to 53, where week 1 is the first week
              that has at least 4 days in the new year.  See also %U and %W.
              (Calculated from tm_year, tm_yday, and tm_wday.)  (SU)

       %W     The week number of the current year as a decimal number, range
              00 to 53, starting with the first Monday as the first day of
              week 01.  (Calculated from tm_yday and tm_wday.)
```

Currently the watch-face uses `%V` (ISO standard) but Outlook uses a different format, which is described here: http://stackoverflow.com/a/2202109/556141

Is there any chance to add it?

Thanks!

tilden, 2016-10-03:
This could definitely be an option, and implemented as a toggle in settings that is only shown when week number is selected.

Muffo, 2016-10-03:
Hi @tilden, that would be great!

Muffo, 2016-10-08:
I started to look at the code and I could not find the source code of the settings page that is displayed on the smartphone.
Is it stored in a separate repository or generated dynamically?

I saw a reference to `config_color.html` in the code but the page does not seem to be in this repository.

Thanks!

tilden, 2016-10-08:
The config page is under the gh-pages branch of this repository. (If you want to test it, I recommended cloning into a separate folder and then switching to the gh-pages branch)