freakified/TimeStylePebble #86: Display week number instead of day of week in date complication

mrath, 2016-10-03:
Since we use week numbers a lot at work, and I can pretty much always remember what day of week it is (like I imagine most people can), this would be a nice option for the date complication. 

Format for week number should also be an option, as is already mentioned in #83

I can have a look at both when I have the time (sometime in the next couple of weeks).

tilden, 2016-10-03:
I'm not a big fan of the implied UX here, since it seems to me that having two separate ways of showing a week number is less than ideal. At some point I might revisit the idea with a more general approach of splitting the date widget into smaller sidebar building blocks, but I'm not really sure.

mrath, 2016-10-04:
TBH I don't see what's wrong with adding it. The user can customize how he/she wants the week number displayed, isn't that what this watchface is all about?
I agree that the possibility to show the week number twice in the sidebar is less than ideal, but it's not far from it - and I don't imagine many people will add it twice. :)