freakified/TimeStylePebble #88: sleep information pops up for long periods of time

Aluke333, 2016-10-21:
I don't wear a watch at night so I only want to see the steps information. Yesterday my sleep time of two hours (bogus information) was on my watch face all day.  Today it is gone, I want the ability to only select steps and not steps and sleep.  I am not sure how the two are suppose to work but it seems periodically it gets stuck on showing me sleep information.

padzor, 2016-10-23:
I've seen this as well -- currently the sleep indicator is permanently there on my KS Pebble Time.  A factory reset got rid of it, but it came back.

tilden, 2016-10-23:
Yes, I've been receiving a large number of emails about this issue. I'm currently investigating the cause, but I haven't been able to replicate it myself yet.

fehmicansaglam, 2016-10-24:
I am currently experiencing the same issue. Any solutions?

Aluke333, 2016-10-25:
Is it programed to show sleep for a short  time in the morning and then change to steps? Is there a way to say I  don't use the sleep functionality of Pebble?

Front Dan C

fehmicansaglam, 2016-10-25:
I did a factory reset and the issue seems to be solved at least for now.

fehmicansaglam, 2016-10-26:
I spoke too early. In the morning the bogus sleep information came back.

tilden, 2016-11-03:
In the latest updated I've simply disabled the sleep function altogether. If Pebble fixes the underlying problem I will put it back.

Spitemare, 2016-11-10:
I've done some refactoring that might fix this. Once I verify I'll submit a pull request.

tilden, 2016-11-10:
@Spitemare Sounds good, are you referring specifically to replacing the polling with subscribing to health service notifications?

Spitemare, 2016-11-10:
@tilden Correct. I've written a simple health service that subs to system health events and caches the appropriate values.

Spitemare, 2016-11-12:
So my changes didn't work. I woke up and saw the sleep widget instead of the steps widget. After walking around for about 45 minutes the widget had not changed back to steps. When I exited the watchface via the menu button and then immediately back to it the widget did change back.

What this tells me is that either the OS isn't sending a HealthEventSleepUpdate event when the user wakes up it is sending the event but `health_service_peek_current_activities()` is returning a mask that contains HealthActivitySleep or HealthActivityRestfulSleep. Unless I'm confused about how things work, when an OS fires an event the activities should change.

Why it returns to normal when exiting and reentering the app is also a mystery. It might be that subscribing and unsubscribing to Pebble Health "resets" activities. Whatever is going on this definitely feels like a bug in Pebble Health.

Spitemare, 2016-11-21:
Pull request submitted. It turns out Pebble Health does **not** send a HealthEventSleepUpdate when the user wakes up. To detect wake up you need to listen to HealthEventMovementUpdate and check if the user is sleeping there as well.

plarus, 2017-01-08:
I am currently using the @Spitemare health service modification (61a5e8f448d210092b308328e1a0bf26941455b9)
It works perfectly with @izhidkov fix (b70daa26add54f3bc81deb90bc88bb6ac1014399)

I have done some execution statistics on this code. Health callback is called about (on average):
    * 1 time per minutes during sleep
    * 4 times per minutes during day (staying indoors / without sport activity)

So it is not very different than constantly asking the system...
I wonder if it is really pertinant to redraw all the display every health event. I am doing some test setting "s_health_event_callback" callback to NULL. Display is refreshed only one time per minute. It works correctly for the moment.

plarus, 2017-01-11:
The better correction for the initial problem may be to add 2 standalone sleep and steps widgets. This is already done in those commit: 664301e2d87097e5cf639018428004c5e71ea0f6 21703e95960a2f28db16161fecdfa1bd0bb75fc4

For the health file, I propose an new version without health service subscription but with an update function witch is called every minute. Done in this commit: f1454d35706b4190ae0aabf6a210e175dc88d131

Spitemare, 2017-01-12:
Polling every minute would just end up with the same scenario at the top of the issue. There's a bug in the firmware that affects what health_service_peek_current_activities() returns. If you poll that constantly you'll end up with HealthActivitySleep or HealthActivityRestfulSleep as part of the bitmask until you restart the watchface. Subscribing to health events seems to avoid this sticky bitmask issue.

I have no idea why this is the case but the health service wasn't really designed to be polled every minute. It was designed to push out updates as needed.

tilden, 2017-01-12:
@Spitemare Yes, based on your previous health-service findings I believe the best course of action is to close this issue, as it is a bug in the Pebble firmware, which will never be updated again.

closed 2017-01-12 by tilden

plarus, 2017-01-12:
@Spitemare What is your source to say that "There's a bug in the firmware that affects what health_service_peek_current_activities"? Is it from your tests results?
Did you have reproduce the bug?

tilden, 2017-01-12:
He's referring to the fact that switching to the health service from polling (which he did in his branch) didn't end up affecting the sleep-stuck bug that cropped up around 4.2. You can read more of his findings if you scroll up to his comment on Nov 12.

Spitemare, 2017-01-12:
I did end up writing some test apps to see what was going on. One polled
every minute to see if the user was sleeping. The other was pushed updates.
Just like this bug, the polling app would get stuck. It didn't happen
consistently, though. The app could run a couple days just fine or it could
get stuck the first night.

The polling app didn't show this problem. It did show a weird behavior in
that the system doesn't send a notification when the user wakes up. I'm not
entirely sure if this is a regression or if it never sent such a
notification. There is a workaround if you look at the commit.

plarus, 2017-01-12:
Ok thank you for yours answers!
I will return to health service subscription if I encounter this bug.