freakified/TimeStylePebble #89: Build Issue - Compilation Error

vijayrajanna, 2016-10-31:
Hi,

I am trying to compile this project by directly importing it into CloudPebble.

After the project is imported, a build (with no changes) will result in the following compilation error.
BuildLog - 1
Setting top to                           : /tmp/tmpZoZg7E 
Setting out to                           : /tmp/tmpZoZg7E/build 
Checking for program webpack             : /app/node_modules/.bin/webpack 
Found Pebble SDK for diorite in:         : /app/sdk3/pebble/diorite 
Checking for program gcc,cc              : arm-none-eabi-gcc 
Checking for program ar                  : arm-none-eabi-ar 
Found Pebble SDK for chalk in:           : /app/sdk3/pebble/chalk 
Checking for program gcc,cc              : arm-none-eabi-gcc 
Checking for program ar                  : arm-none-eabi-ar 
Found Pebble SDK for basalt in:          : /app/sdk3/pebble/basalt 
Checking for program gcc,cc              : arm-none-eabi-gcc 
Checking for program ar                  : arm-none-eabi-ar 
'configure' finished successfully (0.099s)
Waf: Entering directory `/tmp/tmpZoZg7E/build'
menuIcon resource 'MENU_ICON' exceeds the maximum allowed dimensions of (25, 25)

I fixed the above error by removing (temporarily) the MENU_ICON.
However, I again run into another compilation error.

../src/c/clock_area.c: In function 'update_clock_area_layer':
../src/c/clock_area.c:110:3: error: implicit declaration of function 'fctx_set_text_size' [-Werror=implicit-function-declaration]
cc1: all warnings being treated as errors
Waf: Leaving directory `/tmp/tmp1aNrvu/build'
Build failed
 -> task in 'diorite/pebble-app.elf' failed (exit status 1): 
	{task 140499153175440: c clock_area.c -> clock_area.c.23.o}
['arm-none-eabi-gcc', '-std=c99', '-mcpu=cortex-m3', '-mthumb', '-ffunction-sections', '-fdata-sections', '-g', '-fPIE', '-Os', '-D_TIME_H_', '-Wall', '-Wextra', '-Werror', '-Wno-unused-parameter', '-Wno-error=unused-function', '-Wno-error=unused-variable', '-I/app/sdk3/pebble/diorite/include', '-I/tmp/tmp1aNrvu/build', '-I/tmp/tmp1aNrvu', '-I/tmp/tmp1aNrvu/build/include', '-I/tmp/tmp1aNrvu/include', '-I/tmp/tmp1aNrvu/build/src', '-I/tmp/tmp1aNrvu/src', '-I/tmp/tmp1aNrvu/build/diorite', '-I/tmp/tmp1aNrvu/diorite', '-I/tmp/tmp1aNrvu/node_modules/pebble-fctx/dist/include', '-I/tmp/tmp1aNrvu/node_modules/pebble-fctx/dist/include/pebble-fctx/diorite', '-I/tmp/tmp1aNrvu/node_modules/pebble-utf8/dist/include', '-I/tmp/tmp1aNrvu/node_modules/pebble-utf8/dist/include/pebble-utf8/diorite', '-DRELEASE', '-DPBL_PLATFORM_DIORITE', '-DPBL_BW', '-DPBL_RECT', '-DPBL_MICROPHONE', '-DPBL_HEALTH', '-DPBL_SMARTSTRAP', '-DPBL_DISPLAY_WIDTH=144', '-DPBL_DISPLAY_HEIGHT=168', '-DPBL_SDK_3', '-D__FILE_NAME__="clock_area.c"', '../src/c/clock_area.c', '-c', '-o', 'src/c/clock_area.c.23.o']

I have attached both the log files.
[Log_1.txt](https://github.com/freakified/TimeStylePebble/files/562420/Log_1.txt)
[Log_2.txt](https://github.com/freakified/TimeStylePebble/files/562421/Log_2.txt)

Please let me know if I am missing some details.
I wanted to work on some of the feature requests.

Thanks.
Vijay Rajanna.

tilden, 2016-10-31:
These build errors were just introduced in a newer SDK. I already have them fixed-ish on the `feature-emery` branch, but that's not quite ready yet

closed 2016-10-31 by tilden

vijayrajanna, 2016-11-01:
I just compiled `feature-emery` branch, it still has those compilation errors.
Is there any way that I can use compilation flags to suppress those errors?
Or, should I use pebble sdk on linux machine ?

tilden, 2016-11-03:
Just tested again and built a new version without issues.

SDK 4.2.1 (latest), MacOS. I've merged the build-related changes into master, so hopefully that helps.

vijayrajanna, 2016-11-03:
Thanks, that's amazing.
I'll first test on cloudpebble, if it doesn't work, I will compile on Linux.
Thanks.

vijayrajanna, 2016-11-03:
Perfect, I just built it on Cloudpebble, and after adding weather API key, it works great.
I did a few customizations with positions of icons, disconnect icon, etc.