freakified/TimeStylePebble #95: Show QuiteTime is Active

jorsmm, 2016-11-24:
In [SDK 4.3](https://developer.pebble.com/sdk/changelogs/4.3/) you can check if **Quite Time** is active. 
> Added [quiet_time_is_active](https://developer.pebble.com/docs/c/User_Interface/Preferences/#quiet_time_is_active) to permit C apps to check if Quiet Time is active on a watch.

Could you please add this property to show the "mute" icon in the top-left corner (like notifications do):
[image]

tilden, 2016-11-24:
FINALLY!!!!!

I will definitely try and get this added as soon as I have time.

jorsmm, 2016-11-24:
Thanks Dan!

EtorArza, 2024-08-14:
Any updates on this? It would be amazing to get a quiet time complication for the sidebar 🙂.

tilden, 2024-08-15:
Ah, I'm surprised I never actually added this! Unfortunately, development on this project has largely ceased, and I don't even have a working Pebble anymore these days...

EtorArza, 2024-08-15:
Understood, thanks for the reply :) I just realized I can just use airplane mode on the watch and I will get a disconnected icon, which mostly does the same.

tilden, 2025-11-01:
Now that pebble is back, I'll see if i can get this added!

ontheair81, 2026-09-20:
Having a mute icon in the top left corner would be a huge improvement! I sometimes miss notifications because I don't realize the watch is still in quiet time.

PeacheyMcKeitch, 2026-10-17:
+1 for this. :) I need to keep my watch silenced during work so would be cool to have an icon.

tilden, 2026-11-03:
Mute icon in the corner has shipped in 8.0.0!

closed 2026-11-03 by tilden