freakified/TimeStylePebble #96: Version in pebble appstore is broken for diorite, building from source works

ashimokawa, 2016-11-28:
When installing the pbw from the appstore, I cannot use the heart rate or health features. When configuring the watchface to show these, there is just an empty space.

When building the source myself and "sideloading" the pbw, it works however as expected.

tilden, 2016-11-28:
Hm, working for me. for me -- Pebble 2 w/ firmware 4.3, appstore version, Android app.

ashimokawa, 2016-11-29:
@tilden 
Weird, I installed the appstore version again and both icons vanished again (without reconfiguring), then installed the self-built version again and they reappeared (also without reconfiguring)

But since I use https://github.com/Freeyourgadget/Gadgetbridge and not the official app, it *might* be my own weird bug, even though I don't have any idea how this can happen.

Feel free to close this without any attention, I do not expect anyone to support Gadgetbridge. ;)

tilden, 2016-11-29:
Yeah, I haven't gotten any other reports of this, and I'm guessing I would be getting a pretty large volume of emails if P2 users couldn't get HRM or steps, so perhaps it's an issue with Gadgetbridge interacting weirdly with the health API.

Closing for now.

closed 2016-11-29 by tilden