freakified/TimeStylePebble #99: Bottom bar implementation

plarus, 2016-12-03:
For information, I am currently working on a bottom bar implementation that can display up to four widgets.
It is still a work in progress.
I do not know if it will be possible to merge this work in the main branch when finished.
Some previews of the current state of my work:

[image]

[image]

tilden, 2016-12-05:
Oh, this is actually pretty cool. I like your approach to the date format, where we just eliminate the date widget altogether and replace it with the text.

As for integrating this, I believe it could be done, but we'd need to implement a switch for it in the settings, and some refactors would probably be necessary to facilitate switching between the two modes.

Visually, I think it's actually a good idea, it just will need a few refinements. Here's what I'm kinda thinking:
[image]
This centers the time horizontally, centers the colon, and vertically aligns the text and icons with each other. In terms of implementation, I'm not sure how we could achieve that last item, but I believe it could be done.

tilden, 2016-12-06:
So I was thinking more about how we could implement the evenly-aligned bottom bar, and this is the simplest solution I could think of:

We add a second global sidebar control variable, similar to "use compact mode" now, that causes sidebar widgets to draw with the same heights. Not sure how we can get forecast to fit, but maybe that can draw it horizontally or something.

plarus, 2016-12-11:
It is more beautiful with your correction! I have started to implement them.
The new positioning of widgets elements will require some work.
I also have some work updating the html settings page.

After that I expect to allow users to set date fields order and part to be displayed (day of week, day number, month name, month number, year)

tilden, 2016-12-12:
Yeah, I'm thinking that maybe maybe it would be simpler to publish as second watchface entirely, which would simplify the configuration page considerably.

Also, now that Pebble has gone out of business, the app store probably won't have long to live...

plarus, 2016-12-18:
It may just be a revival.
If the pebble store disappear, a replacement apps store may appear ...

I will continue my work on this application.
If you do not went to merge my work (when finished), I will publish it as a new application.

belzo13, 2017-06-13:
Any progress on the bottom bar watchface. I would love to have option of four widgets!

plarus, 2017-06-14:
It is still a work in progress. But very usable by now. Some work are needed in order to have a better config webpage display. I can send you a build version if you want.

belzo13, 2017-06-14:
Would love to have whatever you could offer me. Is a build version ready to install? Thanks!

plarus, 2017-07-11:
You can access to a first pre-release of my work here: https://github.com/plarus/TimeStylePebble/releases

belzo13, 2017-07-12:
Very nice! I like it! Thank You

plarus, 2017-07-16:
Thank you!
If you find an issue you can report it in my own issue list (https://github.com/plarus/TimeStylePebble/issues) as this work is not currently merged in the official version.

Leonick, 2017-10-08:
Hey Plarus,
If you would do something similar for the Round I for one would really appreciate that. Obviously not going to gain space for more complications, still, complications top and bottom with time as large as possible in middle would be really nice.

plarus, 2017-10-15:
Hi Leonick,
Did you expect something like that?

[image]

Leonick, 2017-10-15:
Pretty much, yea. Just make the time larger, the default layout has a few pixels of margin between the time, the widgets, and screen edge.

The widgets are off centre, but I assume that’s just on account of it being WIP.

plarus, 2017-10-15:
You can access to my work here: https://github.com/plarus/TimeStylePebble/releases/tag/0.8.1

Leonick, 2017-11-06:
Been meaning to get around to this, but kept forgetting...

The changes are working great, loving the watchface. Thank you.

plarus, 2017-11-11:
I have released a new version with various corrections and aplite compatibility.

You can download it here : https://github.com/plarus/TimeStyleBBPebble/releases/tag/0.9.0

tilden, 2025-09-14:
@PeacheyMcKeitch I updated TimeStyle recently! It now has a UV Index widget and slightly better weather! Some other new features coming also!

Zatker, 2026-10-28:
Seconding support for bottom/top bar. Currently using the timestylebb branch cuz I love that view and the additional widget space but would be nice to have the added features and updates from here.

tilden, 2026-10-31:
It's coming soon! Just need to do some more testing to ensure it's ready for release!

tilden, 2026-11-03:
Released in 8.0.0 -- ultimately I went for a different design than Plarus did, but I hope people enjoy this version also!