freakified/TimeStylePebble #103: Execution statistics: battery_state_service and Sidebar_redraw

plarus, 2017-01-07:
I am currently doing some execution statistics on timestyle watchface.

Here are my 2 first conclusions:

* "battery_state_service" is called every minutes. So screen display is refreshed twice a minute (and not only one time as expected).
As charge percentage is refreshed only 10% by 10% (on my Pebble 2) it do not looks like pertinent to suscribe to "battery_state_service" for displaying battery charge. Screen refresh occurs every minute and it need some hours to consume 10% battery.
Is there a real reason do suscribe to "battery_state_service"?

* Only calling "Sidebar_redraw" (with it "layer_mark_dirty(sidebarLayer)" call) refresh all the watch screen and not only sidebar. So "update_clock_area_layer" function is also called.
This is may be due to the fact that "sidebarLayer" bounds is included inside  "clock_area_layer".
It look like pertinent to find a method to refresh only the sidebar/clock area part of the screen.
Did you have any idea?

plarus, 2017-01-08:
I have made progress on my second point.
When we have 2 layers in the same window, all layer are always refreshed.
It is needed to create 2 separate window to refresh only one part of the screen.
The ["classio-battery-connection-Economy"](https://github.com/ron064/classio-battery-connection-Economy) watchface use this concept.
But I do not know if this has a real interest... May be only if the "seconds widget" is displayed.

tilden, 2017-01-08:
Do you believe these issues are having any meaningful impact on battery life?

plarus, 2017-01-08:
I do not know the exact impact on power consumption.
My only reference on Pebble watchface power consumption is this youtube video from Pebble team...: https://www.youtube.com/watch?v=TS0FPfgxAso
Doing some benchmark may be needed to know the exact power impact. But I do not want to book my Pebble watch for that...

What I have noticed is that with battery, health, bluetooth, ... service subscription, we can refresh screen up to 10 times by minute. So I suppose reducing this value can produce a noticeable impact on power consumption.

I have made some corrections to improve this. Displaying battery level without subscribing to it service looks like to work correctly. About health service subscription, I am still not sure if it is really needed.

As I do not use the "seconds" widget, I do not have any particular interest to do some correction for my second point... The fact that the 2 layers are always refreshed at the same time is just a point that surprised me at the beginning.

tilden, 2017-01-11:
Yeah, I decided to unmerge the health service subscription because it seemed unnecessarily complicated. I figure the way it works is probably fine. If you have no future objections I will close this issue.

plarus, 2017-01-12:
I propose to remove battery service subscription that is an unneeded code in my point of view: 24f16885c2fd927e27f3a01d07dc36f50b8dffe6 
As a result, screen refresh frequency is divided by 2 and the baterry widget still always work as expected

closed 2017-11-11 by plarus