freakified/TimeStylePebble #105: Weather units problem on reconnection

tecufanujacu, 2017-01-19:
I'm using TimeStyle 7.1 and I setted the weather unit in celsius but everytime the pebble disconnect when it reconnect to the phone the the weather unit goes from celsius to fahrenheit.

I'm using a pebbl classic fw 3.12.2 with [Gadgetbridge](https://github.com/Freeyourgadget/Gadgetbridge).

tilden, 2017-01-19:
This is a bug in Gadgetbridge, they've literally hardcoded in support for TimeStyle's requests, which is fairly impressive.

The way the weather system is implemented in TimeStyle itself is that it _always_ requests the weather in celsius, and converts it on the fly if the units are set to Fahrenheit. I'm guessing they're just sending a `SettingUseMetric = false` on reconnect.

closed 2017-01-19 by tilden

danielegobbetti, 2017-01-19:
We implemented the weather support using version 6.9 that is working well for me (code for reference: https://github.com/Freeyourgadget/Gadgetbridge/blob/master/app/src/main/java/nodomain/freeyourgadget/gadgetbridge/service/devices/pebble/AppMessageHandlerTimeStylePebble.java ).
I just upgraded to version 7.1 and I can confirm the issue. Disconnecting and reconnecting isn't even necessary, just leaving and reentering the watchface makes the unit change. Will debug later.

Since I'm commenting here, thanks a lot for this watchface! :-)

tilden, 2017-01-19:
Strange, I don't think there were any changes between 6.9 and 7.1 that adjusted anything related to the weather unit setting. I just double-checked on mine (Pebble Time, Official App) and it appears to stick with the correct unit even when leaving/reentering or disconnecting/reconnecting, so I think the issue must be in Gadgetbridge somewhere.

I'm glad to hear you picked it as one of the supported weather faces in Gadgetbridge! I hope it being open-source helped!

tecufanujacu, 2017-01-20:
If I'm not in error also with the version 6.11 there wasn't problem.

P.S: is there a way to have the version 6.9  and 6.11? From the store I can download only the last version 7.1

P.S. 2: I thanks you both, @danielegobbetti because without GadgetBridge I would not have ever used a Pebble Watch with its spywere official app and @tilden because TimeStyle is my daily watchface from several months and I can't image my pebble without it.

danielegobbetti, 2017-01-20:
@tilden yes the issue is that what previously was "WeatherUseNightIcon" is now "SettingUseMetric", so we were pushing a "0" to the wrong key. I'll refactor our code and get the key values from the JSON, this way it should be upgrade-proof (as long as the key names in json remain the same).

Bottom line @tecufanujacu , next version of Gadgetbridge (0.17.2(!) **not** 0.17.1, since that one is already released and only waiting for fdroid to pick it up) will solve the issue.

The watchface being open source helped immensely (especially because I was using it since months and wanted it to support weather). Actually most of the configuration can be guessed from the js code (as long as it's not obfuscated), but personally speaking I prefer not to deal with closed source apps :)