freakified/TimeStylePebble #106: Pebble needs to downgrade to 4.2

LenxWei, 2017-03-18:
Firmware 4.3 is terribly unstable, as reported as https://forums.pebble.com/t/does-it-possible-downgrade-pebble-watch-firmware-from-4-3-to-4-2/26995 . I have to downgrade to 4.2, but the TimeStyle in the pebble app store is not compatible with 4.2 anymore. Could you provide a version for 4.2 users? Thanks a lot!

tilden, 2017-03-18:
I've got all my older builds, so I could probably just upload them here for those who wish to use older firmware versions.

LenxWei, 2017-03-18:
Thanks! It should work for us.

tjhaygood, 2019-05-31:
Hey, I know this is old, but did you ever post the older builds? 

I tried downloading the files from an older commit but I'm not sure how to compile them. The freeze/crash bug that's present in 4.3 is driving me nuts, and your watchface is just simply the best one out there.

tilden, 2019-05-31:
Oh dang I completely forgot about this! I will try to dig up the archive and post it.

tilden, 2019-06-09:
Done! All previous versions are now posted here:

https://github.com/freakified/TimeStylePebble/tree/master/Releases

I believe the last version that would work on 4.2 would be this one, 6.10:
https://github.com/freakified/TimeStylePebble/blob/master/Releases/TimeStyle-Release-6.10.pbw

Sorry for the delay, hope that helps!

closed 2019-06-09 by tilden