freakified/TimeStylePebble #110: Port to Fitbit OS?

yellowjaguar5, 2018-01-29:
Hello!
I was wondering if anyone working on this project has plans to port this (amazing) watchface to Fitbit OS (for Fitbit Iconic). I'm thinking about getting an Iconic now that Pebble services are shutting down in 6 months, but I'll hold onto my pebble as long as this watchface is on it!

theLMGN, 2018-12-08:
I was ctually planning on this, then I realized I do know know the FitBit API

arivazhagan, 2019-06-11:
Any update on this? i found a similar watch face https://github.com/TranquilMarmot/StyleTime

tilden, 2019-06-12:
Oh! Not a fan of that one, some really strange design decisions there.

That said, I did start on one for Ionic, but I never finished it. I'll dig up the source code and post it tonight!

tilden, 2019-06-13:
I've pushed the latest code to https://github.com/freakified/TimeStyleFitbit

If anyone would like to take over development, I'd be happy to collaborate! Unfortunately, I don't really have time to work on it anymore.

It currently runs on Ionic, but has the following limitations:

- Config page doesn't actually change watchface colors
- Widgets cannot be changed or reordered
- No weather support
- No Versa support

Here are some of my mockups for the design:

### Ionic:
[image]
[image]

### Versa:
[image]

closed 2020-05-18 by yellowjaguar5

derekwilson, 2020-06-04:
Maybe have a look at this watchface - it should work on all versions of Fitbit Ionic and Versa?

https://gallery.fitbit.com/details/dfe5fccd-01e5-4979-a5ad-070673df12dd

I think it does have its moments - granted I am bias :-)

yellowjaguar5, 2020-06-04:
Very nice. Thanks for sharing @derekwilson.