freakified/TimeStylePebble #116: Settings page 404s

ToxicFrog, 2020-09-30:
Tapping the gear in the Pebble app brings up a "404 - there isn't a github pages site here" error page. Unfortunately I don't know which github page it's trying to load because it doesn't have the usual browser widgets.

Coollision, 2020-09-30:
it tries to load this page http://freakified.github.io/TimeStylePebble/ see https://github.com/freakified/TimeStylePebble/blob/master/src/pkjs/index.js you can find the base config URL there

Bungeetaco, 2020-10-04:
Any plan to fix this, or is there a way to load the page alternatively? I still use this watchface on a daily basis.

plarus, 2020-10-04:
As a workaround solution there is my forked version [TimeStyleBBPebble](https://github.com/plarus/TimeStyleBBPebble) for which the setting page is still working : https://plarus.github.io/TimeStyleBBPebble/config_color.html?appversion=11

theLMGN, 2020-10-21:
Weird. There have been no commits to [`gh-pages`](https://github.com/freakified/TimeStylePebble/tree/gh-pages) since Feb 2017. This seems like an easy fix, @freakified just needs to check the GitHub Pages settings.

tilden, 2020-10-21:
@theLMGN I couldn't find anything wrong in the settings, but I just made a 1-line meaningless change to force a gh-pages rebuild, and it appears that this has fixed it -- http://freakified.github.io/TimeStylePebble/config_color.html now works. 

That said, these days I mostly just recommend @plarus's fork (or one of the other forks that I think exists).

ToxicFrog, 2020-11-06:
I tried @plarus's fork, but the "send to watch" button doesn't do anything, it just remains stuck at the default settings forever. So, thanks for the quick fix!

closed 2025-08-02 by tilden