freakified/TimeStylePebble #124: Watch face settings page don't open.

Arman11, 2023-12-25:
Hey bro! Thanks for the coolest pebble watch face! For some reason, the watch face settings do not open, it refers to a broken page on Github. Please help.
[image]

smilet, 2023-12-25:
I second this, hope it will be back 🤞. 
Timestyle has followed my for years, on multiple watches. Also, let me know where to donate.

Arman11, 2023-12-26:
Found a temporary solution. Follow the link “read the full documentation”, then to the blog, then to the forum, then to your profile on Github. And wherever you need to pin the link: http://freakified.github.io/TimeStylePebble/config_diorite.html?appversion=8. It worked for me :D :D :D

tilden, 2023-12-26:
Huh, that's weird, all the links still seem to work, so I'm not sure what's going on there:

OG Pebble: 
https://freakified.github.io/TimeStylePebble/config_bw.html?appversion=8

Pebble Time: 
https://freakified.github.io/TimeStylePebble/config_color.html?appversion=8

Pebble Time Round:
https://freakified.github.io/TimeStylePebble/config_color_round.html?appversion=8

Pebble 2:
https://freakified.github.io/TimeStylePebble/config_diorite.html?appversion=8

smilet, 2024-01-03:
Hard to say. Something has changed though, it no more loads a 404, now it shows a blank page, no url, just a white blank screen.

CaidenBlock, 2024-01-06:
According to the src it might be trying to open a http link still, I noticed from some android security patch (october 2023?) opening a web view of non-https content can give a blank white screen.

This doesn't explain why it was initially leading to a GitHub 404 page, so it might be unrelated.

I don't have a setup for building pebble apps atm though, so I'll try to test this later

smilet, 2024-01-17:
For me, its still only Timestyle, no other watchface seem to have a broken (blank) settings page.

peekaboo789456, 2024-01-21:
Love this watchface, but can't access settings like others. I'm using iOS and Rebble.  If I try to access the watchface settings I get a black page with a message to check my internet settings and retry.  If I use the links in the reply above for my Pebble Time, I do get a settings menu, but it has a warning that my watchface is an old version, and when trying to apply the settings it won't update.  This is the only watchface I have that does this.  I did find a watchface likely forked from this called "Timestyle BB" that has settings that work.  Still would love to use this one though.

tilden, 2025-10-31:
I believe this should be fixed as of 2025, with the new config page changes.

closed 2025-10-31 by tilden